unitreerobotics/unitree_sim_isaaclab #6: 成功启动命令后，如何触发任务发生

SunJiayu-L, 2025-07-15:
(unitree_sim_env) sun@sun-MS-7D48:~/unitree/unitree_sim_isaaclab$ lsb_release -a 
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.5 LTS
Release:	22.04
Codename:	jammy

+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 570.169                Driver Version: 570.169        CUDA Version: 12.8     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4060        Off |   00000000:01:00.0  On |                  N/A |
|  0%   42C    P3            N/A  /  115W |    1515MiB /   8188MiB |      5%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A             968      G   /usr/lib/xorg/Xorg                      681MiB |
|    0   N/A  N/A            1302      G   /usr/bin/gnome-shell                    161MiB |
|    0   N/A  N/A            3510      G   /opt/google/chrome/chrome                 2MiB |
|    0   N/A  N/A            3558      G   ...ersion=20250714-050047.266000        410MiB |
|    0   N/A  N/A           19230      G   ...ess --variations-seed-version         37MiB |
|    0   N/A  N/A           29359      G   /proc/self/exe                          164MiB |
+-----------------------------------------------------------------------------------------+

我成功启动了 `python sim_main.py --device cpu  --enable_cameras  --task Isaac-PickPlace-Cylinder-G129-Dex3-Joint    --enable_dex3_dds --robot_type g129`  之后，我尝试修改修改圆柱位置为（0.1,0,0)发生了报错

2025-07-15 07:46:48 [103,537ms] [Warning] [omni.physx.tensors.plugin] prim '/World/envs/env_0/Object/geometry/mesh' was deleted while being used by a shape in a tensor view class. The physics.tensors simulationView was invalidated.
2025-07-15 07:46:48 [103,589ms] [Error] [omni.physx.tensors.plugin] Simulation view object is invalidated and cannot be used again to call getDofVelocities

program exception: Failed to get DOF velocities from backend

clean up resources...
[DDSActionProvider] ActionProvider stop
[DDSActionProvider] ActionProvider stopped
[SimpleController] the controller is stopped
[G1RobotDDS] Stopping node...
[G1RobotDDS] Subscribe thread stopped
[G1RobotDDS] Publish thread stopped
[G1RobotDDS] Node stopped
[G1RobotDDS] G1 robot DDS communication stopped
[GripperDDS] Stopping node...
[GripperDDS] Subscribe thread stopped
[GripperDDS] Publish thread stopped
[GripperDDS] Node stopped
[GripperDDS] Gripper DDS communication stopped
cleanup completed
Performing final cleanup...
Current main process PID: 31548
Found related processes: ['31548', '31899']
Terminating child process: 31899
2025-07-15 07:46:50 [105,167ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_OUT_OF_HOST_MEMORY
2025-07-15 07:46:50 [105,170ms] [Error] [carb.graphics-vulkan.plugin] vkCreateFence failed with ERROR_OUT_OF_HOST_MEMORY (file descriptors: 307, soft limit: 2450, hard limit: 1048576).
Force killing process: 31899
2025-07-15 07:46:52 [107,290ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
2025-07-15 07:46:52 [107,290ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
2025-07-15 07:46:52 [107,290ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
2025-07-15 07:46:52 [107,290ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
2025-07-15 07:46:52 [107,394ms] [Warning] [omni.kit.menu.utils.app_menu] add_menu_items: menu [<MenuItemDescription name:'New'>, <MenuItemDescription name:'Open'>, <MenuItemDescription name:'Re-open with New Edit Layer'>, <MenuItemDescription name:'Save'>, <MenuItemDescription name:'Save With Options'>, <MenuItemDescription name:'Save As...'>, <MenuItemDescription name:'Save Flattened As...'>, <MenuItemDescription name:'Add Reference'>, <MenuItemDescription name:'Add Payload'>, <MenuItemDescription name:'Exit'>] cannot change delegate
/home/sun/miniconda3/envs/unitree_sim_env/lib/python3.10/multiprocessing/resource_tracker.py:224: UserWarning: resource_tracker: There appear to be 7 leaked shared_memory objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '

le-wei, 2025-07-15:
你好你是否在图像中的位置修改了位置，我进行同样的设置并且使用同样的命令进行测试，是可以正常启动的。请问除了修改这里有没有修改其他地方的代码，能否给一个更加详细的输出日志。谢谢

[image]

le-wei, 2025-07-15:
命令启动起来以后在界面中点击就可以进行启动任务

SunJiayu-L, 2025-07-16:
对不起，我没有表述清楚意思。我的意思是在isaacsim成功启动之后，我拖拽圆柱体移动位置，期望看见灵巧手跟踪并抓取圆柱体到某个特定位置。 但是发现会发生上述报错。  向请问一下这个任务具体的执行情况是我预计的吗？

le-wei, 2025-07-16:
不好意思，这个项目目前主要通过遥操作进行数据采集，然后通过采集的数据集进行数据回放，并可以通过改变光照条件相机参数等进行数据的生成。项目并没有自动跟踪物体位姿进行抓取功能。如果想调整物体的位置可以通过reset_pose_test.py发送相关信息进行。直接拖拽物体是有可能出问题的。

SunJiayu-L, 2025-07-16:
好的好的，非常感谢。

weihaosky, 2025-07-24:
那请问如何让机器人动起来，我这个现实的是不是缺了一张桌子。。。圆柱体在不停地上下移动，这正常吗

[image]

weihaosky, 2025-07-24:
> <img alt="Image" width="2000" height="2000" src="https://private-user-images.githubusercontent.com/144251895/470228397-8dafbfa2-7d11-45b4-8eb5-2cea71092d1f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTMzNTA5ODgsIm5iZiI6MTc1MzM1MDY4OCwicGF0aCI6Ii8xNDQyNTE4OTUvNDcwMjI4Mzk3LThkYWZiZmEyLTdkMTEtNDViNC04ZWI1LTJjZWE3MTA5MmQxZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcyNFQwOTUxMjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yODAwYmZjNjU5ZWQyOTgyYmIwMTNmYmUyMmU5YWJkZmRjNWQzZDkzNTk5Mzg5MGFkZTE5NjBlOWYzMDE5MGQ3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.SIHJrVxe_O8MJV3XZDfrmRNMEMAEq_sjQkU21FFj6Sc"> 我的是这样的。感觉更像是加载问题。可以重新启动。机器人运动可以通过dds，获取遥操完成。

感谢回复，请问摇操是需要有实体摇操设备吗，我没有实体设备，只有一台电脑

SunJiayu-L, 2025-07-24:
https://github.com/unitreerobotics/unitree_rl_lab  https://github.com/unitreerobotics/unitree_rl_gym