unitreerobotics/unitree_sim_isaaclab #12: how to make camera_base_link

eastflag, 2025-07-23:
I am constructing a task that connects the inspire hand to g1.
The urdf on unitree github doesn't have left_camera_base_link and right_camera_base_link.
What should I do? Is it possible to work on usd? Or should I make urdf and import it?

le-wei, 2025-07-23:
@eastflag We have added support for the Inspire hand. However, due to some missing materials in its URDF, we are unable to fully restore its original colors. Thank you.

eastflag, 2025-07-24:
I was working on inspire dds but you already committed that job. Thank for that.
I have two questions.
First.
I once made a scenario with nvidia CloudXR.
How can I reset a scenario after completing it or failing? In CloudXr, if I say reset(or play or stop), the screen is reset.
Of course, it automatically resets when it succeeds.
Second.
I was able to create 10 scenarios and augmented 1000 data with one command.
Is it possible to augment with one command here too?
I think --modify-light or --modify-camera is like Groott-Gen, I hope you can tell me how to use it in detail.