unitreerobotics/unitree_sim_isaaclab #14: [Bug] Fail when run `Isaac-PickPlace-RedBlock-*` related task

keli-wen, 2025-07-29:
### Description

I can successfully run:

```shell
python sim_main.py --device cpu  --enable_cameras  --task  Isaac-PickPlace-Cylinder-G129-Dex1-Joint    --enable_gripper_dds --robot_type g129
```

But when I tried to run other commands, I encountered error messages.

```shell
python sim_main.py --device cpu  --enable_cameras  --task Isaac-PickPlace-RedBlock-G129-Dex1-Joint    --enable_gripper_dds --robot_type g129
```

The specific errors are as follows:

```shell
...
2025-07-29 11:42:33 [35,364ms] [Error] [rtx.neuraylib.plugin] [MDLC:COMPILER]   1.0   MDLC   comp error: file:/home/keliwen/unitree_sim_isaaclab/assets/objects/small_warehouse_digital_twin/SubUSDs/materials/MI_CeilingA_06b.mdl(56,5): C109 'OmniUe4Base_1' has not been declared; did you mean 'OmniUe4Base'?
2025-07-29 11:42:33 [35,364ms] [Warning] [rtx.neuraylib.plugin] Loading MdlModule to DB (OmniUsdMdl) failed: ::Z73file_3A::home::keliwen::unitree_sim_isaaclab::assets::objects::small_warehouse_digital_twin::SubUSDs::materials::MI_CeilingA_06b
2025-07-29 11:42:33 [35,364ms] [Warning] [rtx.neuraylib.plugin] Loading transaction committed (this thread). MdlModule is NOT in the DB (OmniUsdMdl): ::Z73file_3A::home::keliwen::unitree_sim_isaaclab::assets::objects::small_warehouse_digital_twin::SubUSDs::materials::MI_CeilingA_06b
2025-07-29 11:42:33 [35,364ms] [Error] [omni.usd] USD_MDL: in LoadModule at line 242 of ../../source/plugins/omni.usdMdl/neuray.cpp -- 'rtx::neuraylib::MdlModuleId' for '/home/keliwen/unitree_sim_isaaclab/assets/objects/small_warehouse_digital_twin/SubUSDs/materials/MI_CeilingA_06b.mdl' is Invalid

2025-07-29 11:42:33 [35,365ms] [Error] [omni.usd] USD_MDL: in GetSdrFromDiscoveryResult at line 183 of ../../source/plugins/omni.usdMdl/moduleRegistry.cpp -- Module: '/home/keliwen/unitree_sim_isaaclab/assets/objects/small_warehouse_digital_twin/SubUSDs/materials/MI_CeilingA_06b.mdl' with version '1' not found in 'MdlModuleRegistry::ModuleDataMap'.

2025-07-29 11:42:33 [35,417ms] [Error] [rtx.neuraylib.plugin] [MDLC:COMPILER]   1.0   MDLC   comp error: file:/home/keliwen/unitree_sim_isaaclab/assets/objects/small_warehouse_digital_twin/SubUSDs/materials/MI_CeilingA_06b.mdl(56,5): C109 'OmniUe4Base_1' has not been declared; did you mean 'OmniUe4Base'?
2025-07-29 11:42:33 [35,417ms] [Warning] [rtx.neuraylib.plugin] Loading MdlModule to DB (rtx_scope0) failed: ::Z73file_3A::home::keliwen::unitree_sim_isaaclab::assets::objects::small_warehouse_digital_twin::SubUSDs::materials::MI_CeilingA_06b
2025-07-29 11:42:33 [35,418ms] [Warning] [rtx.neuraylib.plugin] Loading transaction committed (this thread). MdlModule is NOT in the DB (rtx_scope0): ::Z73file_3A::home::keliwen::unitree_sim_isaaclab::assets::objects::small_warehouse_digital_twin::SubUSDs::materials::MI_CeilingA_06b
2025-07-29 11:42:33 [35,418ms] [Error] [omni.hydra] Failed to create MDL shade node for prim '/World/envs/env_0/Room/Assets/rack/shelf/Looks/MI_RackSetC_01/MI_CeilingA_06b'. createMdlModule failed.
2025-07-29 11:42:33 [35,429ms] [Error] [rtx.neuraylib.plugin] [MDLC:COMPILER]   1.0   MDLC   comp error: file:/home/keliwen/unitree_sim_isaaclab/assets/objects/small_warehouse_digital_twin/SubUSDs/materials/MI_CeilingA_06b.mdl(56,5): C109 'OmniUe4Base_1' has not been declared; did you mean 'OmniUe4Base'?
2025-07-29 11:42:33 [35,429ms] [Warning] [rtx.neuraylib.plugin] Loading MdlModule to DB (rtx_scope0) failed: ::Z73file_3A::home::keliwen::unitree_sim_isaaclab::assets::objects::small_warehouse_digital_twin::SubUSDs::materials::MI_CeilingA_06b
2025-07-29 11:42:33 [35,429ms] [Warning] [rtx.neuraylib.plugin] Loading transaction committed (this thread). MdlModule is NOT in the DB (rtx_scope0): ::Z73file_3A::home::keliwen::unitree_sim_isaaclab::assets::objects::small_warehouse_digital_twin::SubUSDs::materials::MI_CeilingA_06b
2025-07-29 11:42:33 [35,429ms] [Error] [omni.hydra] Failed to create MDL shade node for prim '/World/envs/env_0/Room/Assets/rack/frame/Looks/MI_RackSetB_01/MI_CeilingA_06b'. createMdlModule failed.
2025-07-29 11:42:33 [35,438ms] [Error] [rtx.neuraylib.plugin] [MDLC:COMPILER]   1.0   MDLC   comp error: file:/home/keliwen/unitree_sim_isaaclab/assets/objects/small_warehouse_digital_twin/SubUSDs/materials/MI_CeilingA_06b.mdl(56,5): C109 'OmniUe4Base_1' has not been declared; did you mean 'OmniUe4Base'
...
```

I am somewhat confused about this issue because files like `OmniUe4Base_1` do exist:

```shell
$ ls assets/objects/small_warehouse/SubUSDs/materials
Aluminum_Anodized_Black.mdl  Clear_Glass.mdl        MI_DoorA.mdl     M_TrafficCone.mdl  OmniUe4Function.mdl    Paint_Satin.mdl      Plywood.mdl
Aluminum_Polished.mdl        Concrete_Polished.mdl  MI_PC_ON_02.mdl  OmniUe4Base.mdl    OmniUe4Function_1.mdl  Plastic.mdl
Chain_Link_Fence.mdl         MI_Clock.mdl           MI_Props.mdl     OmniUe4Base_1.mdl  Paint_Gloss.mdl        Plastic_Acrylic.mdl
```

le-wei, 2025-07-29:
@keli-wen 
Could you please let us know the environment and system version you are using, as well as your GPU model? This information will help us identify the issue. Also, may I ask which version of Isaac Sim you are currently using? At the moment, our project only supports version 4.5.0.

le-wei, 2025-07-30:
@keli-wen 
Based on the information you provided, it looks like the simulation has already started successfully. The error messages you're seeing are likely due to some dependencies missing from the modified scene we provided, but they do not affect functionality. You can simply left-click on the Sim window to activate rendering. Then, follow the instructions in the README to locate the main camera view. Thank you!

keli-wen, 2025-07-30:
Thank u @le-wei ,  I successfully trigger the simulation env with the README instruction (camera).

closed 2025-07-30 by keli-wen