unitreerobotics/unitree_sim_isaaclab #17: stiffness and damping value do not apply.

eastflag, 2025-08-04:
When I simulated the dataset learned from teleopration, the robot did not move well. 
So, when I checked the robot's setting value like stiffness and damping, those values was not applied.
The original setting that exists in USD file is not changing.

le-wei, 2025-08-04:
@eastflag 
The table below contains the values I obtained from the log file.
I also printed out the actual values after running the program, for your reference.

[image]
🔍 Trying common parameter names:
   ✅ Found stiffness: default_joint_stiffness
      Type: <class 'torch.Tensor'>
      Shape: torch.Size([1, 33])
      Value: tensor([[  100.,   100., 10000.,   100.,   100., 10000.,   100.,   100., 10000.,
           100.,   100.,    25.,    25.,   100.,   100.,    25.,    25.,   100.,
           100.,    25.,    25.,    50.,    50.,    40.,    40.,    40.,    40.,
            40.,    40.,   800.,   800.,   800.,   800.]])
   ✅ Found damping: default_joint_damping
      Type: <class 'torch.Tensor'>
      Shape: torch.Size([1, 33])
      Value: tensor([[1.0000e+00, 1.0000e+00, 1.0000e+04, 1.0000e+00, 1.0000e+00, 1.0000e+04,
         1.0000e+00, 1.0000e+00, 1.0000e+04, 1.0000e+00, 1.0000e+00, 2.0000e+00,
         2.0000e+00, 1.0000e+00, 1.0000e+00, 2.0000e+00, 2.0000e+00, 1.0000e+00,
         1.0000e+00, 2.0000e+00, 2.0000e+00, 2.0000e+00, 2.0000e+00, 2.0000e+00,
         2.0000e+00, 2.0000e+00, 2.0000e+00, 2.0000e+00, 2.0000e+00, 3.0000e+00,
         3.0000e+00, 3.0000e+00, 3.0000e+00]])

eastflag, 2025-08-04:
You mean that   the value is applied when the actual command is sent, even if the values of isaac sim is not changed.
Can you tell me how you printed stiffness and damping values.
In ROS2, values of kp and kd can be found for LowCmd, and the results are as follows.
(.venv) ubuntu@eastflag-desktop:~/ros/ros2_ws$ ros2 interface show  unitree_hg/msg/LowCmd
---
mode_pr: 0
mode_machine: 0
motor_cmd:
- mode: 1
  q: 0.010055800899863243
  dq: 0.0
  tau: 0.0
  kp: 300.0
  kd: 3.0
  reserve: 0
...
But the values are slightly different.

le-wei, 2025-08-05:
@eastflag 
You can use the following code to obtain the relevant `kp`/`kd` values:

```
def get_robot_stiffness_from_env(env):
    """
    Get robot joint stiffness/damping parameters from runtime environment
    Access through env.scene["robot"].data
    """
    print("🔍 Getting joint kp/kd parameters from runtime environment...")
    
    try:
        if not hasattr(env, 'scene'):
            print("❌ Environment has no 'scene' attribute")
            return None
        
        # Display available scene entities
        try:
            scene_keys = list(env.scene.keys()) if hasattr(env.scene, 'keys') else "Unable to get"
            print(f"📋 Available entities in scene: {scene_keys}")
        except:
            print("📋 Unable to get scene entity list")
            
        # Try to access robot object directly
        try:
            robot = env.scene["robot"]
            print(f"✅ Found robot object: {type(robot)}")
        except KeyError as e:
            print(f"❌ Cannot access robot object: {e}")
            return None
        
        if not hasattr(robot, 'data'):
            print("❌ Robot object has no 'data' attribute")
            return None
            
        robot_data = robot.data
        print(f"✅ Found robot.data object: {type(robot_data)}")
        
        # Explore all attributes in robot.data
        print("\n📊 Available attributes in robot.data:")
        print("-" * 60)
        
        data_attrs = [attr for attr in dir(robot_data) if not attr.startswith('_')]
        print(f"📈 Total attributes: {len(data_attrs)}")
        
        stiffness_attrs = []
        damping_attrs = []
        joint_attrs = []
        other_attrs = []
        
        for attr in data_attrs:
            if 'stiff' in attr.lower():
                stiffness_attrs.append(attr)
            elif 'damp' in attr.lower():
                damping_attrs.append(attr)
            elif 'joint' in attr.lower():
                joint_attrs.append(attr)
            else:
                other_attrs.append(attr)
        
        # Display joint-related attributes
        if joint_attrs:
            print(f"\n🔗 Joint-related attributes:")
            for attr in joint_attrs:
                try:
                    value = getattr(robot_data, attr)
                    print(f"   📌 {attr}: {type(value)} - shape: {getattr(value, 'shape', 'N/A')}")
                except Exception as e:
                    print(f"   📌 {attr}: Cannot access - {e}")
        
[... 117 lines cut ...]
                except Exception as e:
                    print(f"   ⚠️ get_dampings() failed: {e}")
        
        # Return results
        result = {
            'stiffness': stiffness_found,
            'damping': damping_found,
            'robot_data': robot_data
        }
        
        print(f"\n✅ Parameter acquisition complete!")
        if stiffness_found is not None or damping_found is not None:
            print("🎉 Successfully found at least one parameter!")
        else:
            print("⚠️ No expected parameters found")
            
        return result
        
    except Exception as e:
        print(f"❌ Error getting parameters: {e}")
        import traceback
        traceback.print_exc()
        return None

```

eastflag, 2025-08-05:
Thank you for your response.
However, I still haven't solved the problem of robots not moving well.
I think the reason is inspire hand. The urdf I had before was 6 joint and 6 mimic joint. And we also have real g1 robot and real inspire hand of 6 joint and 6 mimic joint now.
However, the inspire hand USD in this repository does have 12 joints and does not have mimic joints.

le-wei, 2025-08-05:
@eastflag 
You're right — in the USD file, the Inspire hand does not include simulated joints. However, in `xr_teleoperate`, we only map the joint data from the real moving joints; the rest of the data is computed accordingly. We've created a moving demo, and the result looks like this. Once everything is ready, we’ll upload the update.