unitreerobotics/unitree_sim_isaaclab #18: when I run this example, isaacsim report  MDL error,how can I fix it?

dailouer-guo, 2025-08-06:
[image]

[image]

le-wei, 2025-08-06:
It appears that some materials may not have been downloaded properly. I recommend restarting and trying again.

dailouer-guo, 2025-08-06:
> It appears that some materials may not have been downloaded properly. I recommend restarting and trying again.

Actually, I did , but I still have this error

le-wei, 2025-08-06:
May I ask if you have modified the asset path in Isaac Sim?
In this scene, the assets are expected to be loaded from the `ISAAC_NUCLEUS_DIR` path.
If you have changed this path, it could lead to missing materials.
The default path is:
**`http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac`**.
f"{ISAAC_NUCLEUS_DIR}/Props/PackingTable/packing_table.usd"

dailouer-guo, 2025-08-06:
> May I ask if you have modified the asset path in Isaac Sim? In this scene, the assets are expected to be loaded from the `ISAAC_NUCLEUS_DIR` path. If you have changed this path, it could lead to [...]

i downloaded isaacsim assets and modified the asset path, but I cant found packing_table.usd from this assets, so I asked unitree technical service, he provided me the usd and cfg, it help me started this example successfully

[image]

dailouer-guo, 2025-08-06:
> <img alt="Image" width="2000" height="58" src="https://private-user-images.githubusercontent.com/13995579/474807228-49f125e4-98de-4997-880f-bc23a5a3cfca.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ0NTA3ODUsIm5iZiI6MTc1NDQ1MDQ4NSwicGF0aCI6Ii8xMzk5NTU3OS80NzQ4MDcyMjgtNDlmMTI1ZTQtOThkZS00OTk3LTg4MGYtYmMyM2E1YTNjZmNhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODA2VDAzMjEyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkyY2ZlNDZmOTVjYzQ3ODYzNmQ0NDA0MzgwMzgzNjU3MDMxNzgyNGRjMjg4NTk2MzljMjQ3MGQwYjgxZmUyMWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nRth6CzeKALodHnhyhX-ITnWjaiXVRiYWWysNUbnKxQ"> Could you please verify whether the path shown in the image contains the corresponding material?

[image]