unitreerobotics/unitree_sim_isaaclab #22: how to set rot_offset

eastflag, 2025-08-13:
The right wrist camera turned 90 degrees, so I rightly rotated it to (3.999, -58, -102.0)
And its quternion is w, x, y, z = (0.53694, 0.39574, -0.2812, -0.678994) in Isaac Sim

[image]

I inserted these values to camera_config.py.
    @classmethod
    def right_inspire_wrist_camera(cls) -> CameraCfg:
        """right wrist camera configuration"""
        return CameraBaseCfg.get_camera_config(
...
            rot_offset=(0.39574, -0.2812, -0.678994, 0.53694),  # x, y, z, w because convention is "ros"
        ) 

But after running simulator, the result is not the same as I expect.
How do I have to set this values?

le-wei, 2025-08-13:
The order for setting the camera pose is (-x, w, z, -y).

eastflag, 2025-08-13:
I used this.
rot_offset=(-0.39574, 0.53694, -0.678994, 0.2812)
and result

[image]

this is what I want. Thanks.

closed 2025-08-14 by eastflag