unitreerobotics/unitree_sim_isaaclab #23: Use of the arm_sdk dds channel and Loco Client

nitish3693, 2025-08-14:
Hi Team, 

I am trying to use the high level examples G1 from the https://github.com/unitreerobotics/unitree_sdk2_python/tree/master/example/g1/high_level . When i start the [g1_arm7_sdk_dds_example.py](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/example/g1/high_level/g1_arm7_sdk_dds_example.py) the robot makes no action. 

I was starting the scene with the comman, 
`python sim_main.py --device cpu  --enable_cameras  --task Isaac-Move-Cylinder-G129-Dex3-Wholebody  --enable_dex3_dds --enable_wholebody_dds --robot_type g129`

Outputs : 
```
***  Please left-click on the Sim window to activate rendering. ***

2025-08-14 13:50:46 [17,015ms] [Warning] [omni.physx.plugin] PhysicsUSD: PhysxMaterialAPI at prim /World/envs/env_0/Room/Assets/dollies/dolly/wheelMaterial has attribute "improvePatchFriction" set to false. This setting is deprecated and support will end soon.
2025-08-14 13:50:46 [17,092ms] [Warning] [omni.physx.plugin] PhysX warning: PxMaterial::setFlag(): the friction behavior with the flag PxMaterialFlag::eIMPROVED_PATCH_FRICTION cleared is deprecated and support will end soon., FILE /builds/omniverse/physics/physx/source/physx/src/NpMaterial.cpp, LINE 186
2025-08-14 13:50:46 [17,118ms] [Warning] [isaaclab.assets.articulation.articulation] Spatial tendons are not supported in Isaac Sim < 5.0: patching spatial-tendon getter and setter to use dummy value
dds_manager: <dds.dds_master.DDSManager object at 0x7c30b8124cd0>
[DDSManager] object 'g129' not found, objects: dict_keys([])
[g1_state] G1 robot DDS communication instance obtained
g1_robot_dds is not initialized
[DDSManager] object 'dex3' not found, objects: dict_keys([])
[Observations Dex3] DDS communication instance obtained
2025-08-14 13:54:00 [210,861ms] [Warning] [omni.kit.widget.context_menu.custom_menu_dict] _merge_submenu: failed to merge duplicate item None
2025-08-14 13:54:00 [210,862ms] [Warning] [omni.kit.widget.context_menu.custom_menu_dict] _merge_submenu: failed to merge duplicate item Select All
2025-08-14 13:54:00 [210,862ms] [Warning] [omni.kit.widget.context_menu.custom_menu_dict] _merge_submenu: failed to merge duplicate item Select None
2025-08-14 13:54:00 [210,862ms] [Warning] [omni.kit.widget.context_menu.custom_menu_dict] _merge_submenu: failed to merge duplicate item Select Invert
2025-08-14 13:54:00 [210,862ms] [Warning] [omni.kit.widget.context_menu.custom_menu_dict] _merge_submenu: failed to merge duplicate item Select Parent
2025-08-14 13:54:00 [210,862ms] [Warning] [omni.kit.widget.context_menu.custom_menu_dict] _merge_submenu: failed to merge duplicate item Select Leaf
2025-08-14 13:54:00 [210,862ms] [Warning] [omni.kit.widget.context_menu.custom_menu_dict] _merge_submenu: failed to merge duplicate item Select Hierarchy
2025-08-14 13:54:00 [210,862ms] [Warning] [omni.kit.widget.context_menu.custom_menu_dict] _merge_submenu: failed to merge duplicate item Select Similar
2025-08-14 13:54:00 [210,862ms] [Warning] [omni.kit.widget.context_menu.custom_menu_dict] _merge_submenu: failed to merge duplicate item Assembly
2025-08-14 13:54:00 [210,862ms] [Warning] [omni.kit.widget.context_menu.custom_menu_dict] _merge_submenu: failed to merge duplicate item Group
2025-08-14 13:54:00 [210,862ms] [Warning] [omni.kit.widget.context_menu.custom_menu_dict] _merge_submenu: failed to merge duplicate item Component
2025-08-14 13:54:00 [210,862ms] [Warning] [omni.kit.widget.context_menu.custom_menu_dict] _merge_submenu: failed to merge duplicate item Subcomponent
========= create image server =========
[Image Server] Initializing multi-image server from shared memory
[MultiImageReader] Shared memory opened: isaac_multi_image_shm
[Image Server] Multi-image server initialized on port 5555
[Image Server] Starting send_process from shared memory...
[Image Server] Multi-image publishing thread started
========= create image server success =========
========= create dds =========
[g1_robot] Input shared memory: psm_c3d26eb2
[g1_robot] Output shared memory: psm_5533fb6a
[g1_robot] G1 robot DDS node initialized
[DDSManager] register object 'g129' success (category: No category)
[dex3] Input shared memory: psm_45637baf
[dex3] Output shared memory: psm_d395d46c
[dex3] Hand DDS node initialized
[DDSManager] register object 'dex3' success (category: No category)
[run_command_dds] Input shared memory: psm_57078e36
[run_command_dds] Output shared memory: psm_639a928b
[run_command_dds] Run command DDS node initialized
[DDSManager] register object 'run_command' success (category: No category)
[reset_pose_dds] Output shared memory: psm_f698866f
[reset_pose_dds] Reset pose DDS node initialized
[DDSManager] register object 'reset_pose' success (category: No category)
[sim_state_dds] Input shared memory: psm_d316d282
[sim_state_dds] Sim state DDS node initialized
[DDSManager] register object 'sim_state' success (category: No category)
[g1_robot] State publisher initialized (rt/lowstate)
[dex3] Hand state publisher initialized
[sim_state_dds] Sim state publisher initialized
[DDSManager] publish loop thread started
[DDSManager] manager started, managing 5 objects
[g1_robot] Create ChannelSubscriber...
[dex3] Hand command subscriber initialized
[run_command_dds] Run command subscriber initialized
[reset_pose_dds] Reset pose command subscriber initialized
========= create dds success =========

create action provider: dds...
args_cli.task: Isaac-Move-Cylinder-G129-Dex3-Wholebody
ActionProvider init
enable_robot: g129
enable_gripper: False
enable_dex3: True
[DDSActionProvider] DDS communication initialized
========= create controller =========
  - control frequency: 500Hz
[SimpleController] set the action provider: DDSActionProvider
========= create controller success =========
performance analysis enabled, report every 500 steps
Note: The DDS in Sim transmits messages on channel 1. Please ensure that other DDS instances use the same channel for message exchange by setting: ChannelFactoryInitialize(1).
========= start controller =========
[DDSActionProvider] ActionProvider started
[SimpleController] the controller is started
========= start controller success =========
dds_manager: <dds.dds_master.DDSManager object at 0x7c30b8124cd0>
[g1_state] G1 robot DDS communication instance obtained
[Observations Dex3] DDS communication instance obtained

=== While loop execution frequency statistics ===
loop execution count: 96
running time: 10.03 seconds
overall average frequency: 9.57 Hz
moving average frequency: 9.57 Hz (last 96 times)
frequency range: 6.58 - 8371.86 Hz
average loop time: 104.47 ms
recent loop time: 104.47 ms
=============================
```

Upon inspection it appears the dds channel `rt/arm_sdk` is not initialized. 

Could you please suggest if this dds channel is available ? Also can the RPC routine to control the locomotion be used as well.

nitish3693, 2025-08-19:
In the readme it mentions : 

- After the simulator starts running, it will send/receive the same DDS topics as the real robot (Please note to distinguish between the simulator and real robot if there is a real robot running on the same network). For specific DDS usage, please refer to[G1 Control](https://github.com/unitreerobotics/unitree_sdk2_python/tree/master/example/g1) and [Dex3 Dexterous Hand Control](https://github.com/unitreerobotics/unitree_sdk2/blob/main/example/g1/dex3/g1_dex3_example.cpp)

le-wei, 2025-08-20:
@nitish3693 
We sincerely apologize for the misunderstanding. The purpose of the README is to help users who are unfamiliar with DDS by providing example cases to learn how to use it.
The control method you’re trying to use has not been tested in the SIM environment. One thing we can confirm is that the low-level motion control program is not deployed in SIM, so it is indeed different from the real robot.