unitreerobotics/unitree_sim_isaaclab #30: 3070上可以运行吗

JimiStack, 2025-08-21:
能正常运行send_command.py和sim_main.py，但是键盘操作机器人动不了

le-wei, 2025-08-21:
你需要把光标放到执行_command.py的终端上操作

JimiStack, 2025-08-21:
[image]
是这样操作的，但机器人始终不移动

JimiStack, 2025-08-21:
为了验证是不是没监听到键盘信号，加了一行print(f"commands: {commands}")，删掉之后也是这样

[image]

le-wei, 2025-08-21:
可以/home/unitree/Code/unitree_sim_isaaclab/dds/commands_dds.py  中打印一下，测试是否接收到了指令，顺便看一下GPU/CPU占用情况。

le-wei, 2025-08-21:
你先在commands_dds.py订阅消息的函数中打印是否接收到外部指令消息，如果确定是可以正常接收到，那么很可能是硬件的导致的，建议换一个高版本的显卡进行尝试。渲染和模型推理都需要显存，而且isaacsim 有最低配置的要求，[isaacsim](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/index.html)

closed 2025-08-25 by le-wei