unitreerobotics/unitree_sim_isaaclab #31: program exception: 'bool' object has no attribute 'nonzero'

N0r9st, 2025-08-21:
Hello everyone! I cannot get the main teleoperation example to work.

First of all, running this command
```
python sim_main.py --device cpu  --enable_cameras  --task  Isaac-PickPlace-Cylinder-G129-Dex1-Joint    --enable_gripper_dds --robot_type g129

```
crashes because `--enable_gripper_dds` flag is non existing.

And finally, when I fix this, simulator crashes when I left click on window as script asks

```
========================================================

***  Please left-click on the Sim window to activate rendering. ***

dds_manager: <dds.dds_master.DDSManager object at 0x7a68bd329dd0>
[DDSManager] object 'g129' not found, objects: dict_keys([])
[g1_state] G1 robot DDS communication instance obtained
g1_robot_dds is not initialized
[DDSManager] object 'dex1' not found, objects: dict_keys([])
[Observations] DDS communication instance obtained
2025-08-21T15:18:41Z [315,329ms] [Warning] [carb] Client gpu.foundation.plugin has acquired [gpu::unstable::IMemoryBudgetManagerFactory v0.1] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
========= create image server =========
[Image Server] Initializing multi-image server from shared memory
[MultiImageReader] Shared memory opened: isaac_multi_image_shm
[Image Server] Multi-image server initialized on port 5555
[Image Server] Starting send_process from shared memory...
[Image Server] Multi-image publishing thread started
========= create image server success =========
========= create dds =========
[g1_robot] Input shared memory: psm_16ae3bc4
[g1_robot] Output shared memory: psm_56479146
[g1_robot] G1 robot DDS node initialized
[DDSManager] register object 'g129' success (category: No category)
[reset_pose_dds] Output shared memory: psm_92139877
[reset_pose_dds] Reset pose DDS node initialized
[DDSManager] register object 'reset_pose' success (category: No category)
[sim_state_dds] Input shared memory: psm_fafb3e8d
[sim_state_dds] Sim state DDS node initialized
[DDSManager] register object 'sim_state' success (category: No category)
[g1_robot] State publisher initialized (rt/lowstate)
[sim_state_dds] Sim state publisher initialized
[DDSManager] publish loop thread started
[DDSManager] manager started, managing 3 objects
[g1_robot] Create ChannelSubscriber...
[reset_pose_dds] Reset pose command subscriber initialized
========= create dds success =========

create action provider: dds...
args_cli.task: Isaac-PickPlace-Cylinder-G129-Dex1-Joint
ActionProvider init
enable_robot: g129
enable_gripper: False
enable_dex3: False
[DDSActionProvider] DDS communication initialized
========= create controller =========
  - control frequency: 500Hz
[SimpleController] set the action provider: DDSActionProvider
========= create controller success =========
performance analysis enabled, report every 500 steps
Note: The DDS in Sim transmits messages on channel 1. Please ensure that other DDS instances use the same channel for message exchange by setting: ChannelFactoryInitialize(1).
========= start controller =========
[DDSActionProvider] ActionProvider started
[SimpleController] the controller is started
========= start controller success =========

program exception: 'bool' object has no attribute 'nonzero'

clean up resources...
```

Any help would be much appreciated!

le-wei, 2025-08-22:
The error program exception: 'bool' object has no attribute 'nonzero' was caused by IsaacLab.
In my opinion, now is not a good time to switch to the main branch.

eastflag, 2025-08-22:
Sorry,
I got this problem after updating the main branch.
So I reset to the 8/8 commit version.

[image]

closed 2025-08-22 by N0r9st

danielwright-lgtm, 2025-10-24:
Hi,

I am getting this error now. I have IsaacSim 5.0.0 and the main branch of IsaacLab. How can I fix it?

Cheers