unitreerobotics/unitree_sim_isaaclab #43: 渲染延迟

navono, 2025-09-03:
环境：
OS: Ubuntu 22.04
GPU: RTX 4070 Ti Super
CPU: 14代 i7

目前通过文档，能够跑通 Wholebody 的示例，通过 `send_commands_keyboard.py` 也能控制。但是在控制的过程中，能明显感觉到按键按下后到 Sim 中 G1 开始移动有延迟。

GPU 使用情况：

[image]

sim_main 的输出：
```
=== While loop execution frequency statistics ===
loop execution count: 9363
running time: 844.06 seconds
overall average frequency: 11.09 Hz
moving average frequency: 10.38 Hz (last 100 times)
frequency range: 7.75 - 14.09 Hz
average loop time: 90.15 ms
recent loop time: 96.36 ms
=============================

=== While loop execution frequency statistics ===
loop execution count: 9472
running time: 854.13 seconds
overall average frequency: 11.09 Hz
moving average frequency: 10.81 Hz (last 100 times)
frequency range: 7.46 - 16.43 Hz
average loop time: 90.17 ms
recent loop time: 92.49 ms
=============================
[Performance] A:91.6ms, E:0.0ms, S:0.0ms, T:91.6ms

=== While loop execution frequency statistics ===
loop execution count: 9580
running time: 864.22 seconds
overall average frequency: 11.09 Hz
moving average frequency: 10.73 Hz (last 100 times)
frequency range: 7.66 - 14.69 Hz
average loop time: 90.21 ms
recent loop time: 93.24 ms
=============================
```

有什么办法可以改善这种延迟吗？

hyfu0583, 2025-09-03:
same problem
OS: Ubuntu 22.04
GPU: RTX 3090
isaacsim:4.5.0

le-wei, 2025-09-04:
目前占用资源最高的是 Omni 底层部分，可以选择减少软渲染或者简化场景。 我们会在后续持续改进和优化它.

=== While loop execution frequency statistics ===
loop execution count: 452
running time: 20.01 seconds
overall average frequency: 22.59 Hz
moving average frequency: 22.56 Hz (last 100 times)
frequency range: 19.23 - 27.34 Hz
average loop time: 44.27 ms
recent loop time: 44.33 ms
=============================
[Performance] A:49.3ms, E:0.0ms, S:0.0ms, T:49.3ms

=== While loop execution frequency statistics ===
loop execution count: 665
running time: 30.04 seconds
overall average frequency: 22.14 Hz
moving average frequency: 21.03 Hz (last 100 times)
frequency range: 16.15 - 26.03 Hz
average loop time: 45.17 ms
recent loop time: 47.54 ms
=============================

=== While loop execution frequency statistics ===
loop execution count: 879
running time: 40.07 seconds
overall average frequency: 21.94 Hz
moving average frequency: 21.33 Hz (last 100 times)
frequency range: 18.29 - 26.31 Hz
average loop time: 45.58 ms
recent loop time: 46.88 ms
=============================
[Performance] A:44.0ms, E:0.0ms, S:0.0ms, T:44.0ms

=== While loop execution frequency statistics ===
loop execution count: 1091
running time: 50.08 seconds
overall average frequency: 21.78 Hz
moving average frequency: 20.94 Hz (last 100 times)
frequency range: 17.84 - 24.69 Hz
average loop time: 45.90 ms
recent loop time: 47.76 ms
=============================

cleanup completed
         5735401 function calls (5658320 primitive calls) in 85.000 seconds

   Ordered by: internal time
   List reduced from 3173 to 30 due to restriction <30>

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
     1841   58.252    0.032   62.392    0.034 {built-in method omni.kit.app._app.update}
     5990    5.362    0.001    5.386    0.001 {built-in method omni.physx.bindings._physx.fetch_results}
     5989    2.662    0.000    3.500    0.001 articulation.py:1801(_apply_actuator_model)
     5988    1.459    0.000    1.459    0.000 {built-in method omni.physx.bindings._physx.simulate}
    40428    1.073    0.000    2.865    0.000 sensor_base.py:183(update)
     6738    0.909    0.000    1.470    0.000 contact_sensor.py:326(_update_buffers_impl)
        4    0.800    0.200    0.800    0.200 {built-in method time.sleep}
     5992    0.681    0.000    0.723    0.000 articulation_data.py:742(joint_acc)
        2    0.525    0.262    0.525    0.263 {built-in method gc.collect}
    35934    0.500    0.000    0.730    0.000 actuator_pd.py:115(compute)
     3434    0.478    0.000    1.945    0.001 camera.py:493(_update_buffers_impl)
     1839    0.456    0.000    0.456    0.000 {built-in method omni.graph.core._omni_graph_core.evaluate}
   129942    0.386    0.000    0.549    0.000 gestures.py:67(should_prevent)
     3434    0.348    0.000    0.348    0.000 {method 'copy' of 'numpy.ndarray' objects}
    21013    0.294    0.000    0.294    0.000 {method 'clone' of 'torch._C.TensorBase' objects}
    11240    0.278    0.000    3.899    0.000 sensor_base.py:348(_update_outdated_buffers)
     1497    0.271    0.000    0.271    0.000 {built-in method onnxruntime.capi.onnxruntime_pybind11_state.run}
     1837    0.243    0.000    0.243    0.000 {built-in method omni.physxfabric.bindings._physxFabric.force_update}
       22    0.219    0.010    0.243    0.011 {built-in method omni.graph.core._omni_graph_core.create_node}
        2    0.218    0.109    0.218    0.109 {built-in method omni.physx.bindings._physx.force_load_physics_from_usd}
     1497    0.197    0.000   68.932    0.046 action_provider_wh_dds.py:375(get_action)
     6004    0.189    0.000    0.207    0.000 simulation_manager.py:373(get_physics_dt)
     2048    0.181    0.000    0.184    0.000 utils.py:1029(get_local_transform_SRT)
    36054    0.180    0.000    0.224    0.000 frontend_torch.py:75(get_tensor_desc)
     7288    0.176    0.000    0.176    0.000 encoder.py:205(iterencode)
     1497    0.173    0.000    0.580    0.000 action_provider_wh_dds.py:313(compute_current_observations)
     1840    0.163    0.000    0.180    0.000 api.py:465(_conform_projection)
    26952    0.159    0.000    0.159    0.000 {built-in method torch.where}
   144336    0.159    0.000    0.159    0.000 {built-in method omni.ui_scene._scene.should_prevent}
     9735    0.157    0.000    0.157    0.000 {method 'roll' of 'torch._C.TensorBase' objects}