unitreerobotics/unitree_sim_isaaclab #51: Failed to write reset pose command:  program exception: 'DDSRLActionProvider' object has no attribute 'get_start_loop'

QAQ2022, 2025-09-18:
数据回放的时候报错Failed to write reset pose command:

[SimpleController] set the action provider: DDSActionProvider
========= create controller success =========
performance analysis enabled, report every 500 steps
Note: The DDS in Sim transmits messages on channel 1. Please ensure that other DDS instances use the same channel for message exchange by setting: ChannelFactoryInitialize(1).
========= start controller =========
[DDSActionProvider] ActionProvider started
[SimpleController] the controller is started
========= start controller success =========
Failed to write reset pose command:

program exception: 'DDSRLActionProvider' object has no attribute 'get_start_loop'

clean up resources...
[DDSActionProvider] ActionProvider stop
[DDSActionProvider] ActionProvider stopped
[SimpleController] the controller is stopped
cleanup completed
Performing final cleanup...
Current main process PID: 2534844
Found related processes: ['2534844', '2535866']
Terminating child process: 2535866
2025-09-18 10:01:44 [18,918ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_OUT_OF_HOST_MEMORY
2025-09-18 10:01:44 [18,919ms] [Error] [carb.graphics-vulkan.plugin] vkCreateFence failed with ERROR_OUT_OF_HOST_MEMORY (file descriptors: 314, soft limit: 2450, hard limit: 1048576).
Force killing process: 2535866
2025-09-18 10:01:46 [21,015ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
2025-09-18 10:01:46 [21,015ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
2025-09-18 10:01:46 [21,015ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
2025-09-18 10:01:46 [21,015ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
2025-09-18 10:01:46 [21,015ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
2025-09-18 10:01:46 [21,102ms] [Warning] [omni.kit.menu.utils.app_menu] add_menu_items: menu [<MenuItemDescription name:'New'>, <MenuItemDescription name:'Open'>, <MenuItemDescription name:'Re-open with New Edit Layer'>, <MenuItemDescription name:'Save'>, <MenuItemDescription name:'Save With Options'>, <MenuItemDescription name:'Save As...'>, <MenuItemDescription name:'Save Flattened As...'>, <MenuItemDescription name:'Add Reference'>, <MenuItemDescription name:'Add Payload'>, <MenuItemDescription name:'Exit'>] cannot change delegate
/home/jzx/miniconda3/envs/lab/lib/python3.10/multiprocessing/resource_tracker.py:224: UserWarning: resource_tracker: There appear to be 5 leaked shared_memory objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '
我的指令是： python sim_main.py --device cuda  --enable_cameras  --task  Isaac-Move-Cylinder-G129-Dex3-Wholebody    --enable_dex3_dds --robot_type g129 --replay  --file_path "/home/jzx/xr_teleoperate/teleop/utils/data/episode_0002"

QAQ2022, 2025-09-18:
我跟你们说DDSRLActionProvider所在的文件：action_provider_wh_dds.py，，，人家根本就没有get_start_loop这个函数，没用显示调用，，，，隐式的我也看不出来啊问题是

le-wei, 2025-09-18:
sim_state  是用于恢复仿真状态的，没有的话无法正常恢复场景状态

QAQ2022, 2025-09-19:
[image]我保存数据时候这个也不对啊

QAQ2022, 2025-09-19:
8:34:05:620616 INFO     ==> Episode saved start...                                                          episode_writer.py:194
18:34:05:620713 INFO     published reset category: 1                                                     teleop_hand_and_arm.py:33
18:34:06:654006 INFO     ==> Episode saved successfully to ./utils/data/episode_0004/data.json.              episode_writer.py:207

thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Connection(IoError(Custom { kind: Other, error: UnknownError }))'
数据大小问题sim_state在我修改def start_sim_state_subscribe(shm_name: str = "sim_state_cmd_data", shm_size: int = 4096) -> SimStateSubscriber:解决了，默认为3096，，但是数据回放是一样的报错